geospace-code/georinex #1: Reliant on optionary header record

Ydmir, 2016-08-24:
On reading some RINEX files _rinexobs_ fails with the following message "IndexError: list index out of range" after trying to read the number of satellites from the header:
numberOfSv = int(grabfromhead(header,None,6,"# OF SATELLITES")[0][0]

The # OF SATELLITES record is an optional header and it is thus problematic to rely on it.

scienceopen, 2016-08-24:
Thank you for noting this, one of my collaborators @gregstarr noticed the same thing and has a pull request he intends to send fixing this and other issues

scienceopen, 2016-08-29:
@gregstarr pushed his changes to master branch; I didn't have a chance to try them other than to see the self-test still works.

Ydmir, 2017-01-09:
I do not experience the same issues anymore, so it seems to be fixed.

closed 2017-01-09 by Ydmir