geospace-code/georinex #4: BUG: Satellites from multiple systems assigned to same number

Ydmir, 2017-01-10:
Consider the following record header:

17  1  2 13 15  0.0000000  0 30S27G30R24G26R23G16E01R17R01G13R09G09
                                R07G23G29R06G02R08R16G05S28G06G07E09
                                R10E22S36S20S23S26

In it we have amongst other, the satellites R23, G23, and S23, belonging to the GLONASS, GPS, and SBAS constellations respectively. Currently, the code for enumerating satellites in the pandas tables, create the labels from the following:
`sp.append(int(lines[i][33+(s%12)*3:35+(s%12)*3]))`.
This only account for the satellite number, without accounting for its constellation, creating serious errors when trying to use the code on multi-constallation files.

gregstarr, 2017-01-10:
Everything has been fixed in pygps

Check it out

https://github.com/gregstarr/PyGPS

Ydmir, 2017-01-10:
Looking through the PyGPS repo it seems that it is including the code from this repo from a certain time. It seems at it contains the same bug as here. [Line 327](https://github.com/gregstarr/PyGPS/blob/master/gps.py#L327) of the gps.py code contains the exact same code as I referenced in this issue.

gregstarr, 2017-01-10:
Okay I know I fixed that issue somewhere,  it could possibly be here

https://github.com/aldebaran1/gsit

Or it's possible I haven't updated the online version yet.

Sorry for the inconvenience, i will be home in about a week and will fix it
then

Ydmir, 2017-02-01:
Any updates on this issue?

gregstarr, 2017-02-01:
Pyrinex doesn't currently support glonass

gregstarr, 2017-02-02:
pyrinex and PyGPS don't currently support glonass

scivision, 2018-05-08:
All (many?) satellite types are handled by recent versions of PyRinex, which uses xarray interface to remove ambiguity over satellite numbers.

closed 2018-05-08 by scivision