geospace-code/georinex #6: cast to int an unicode variable

Cami23, 2017-10-20:
This codeline, i can't compile in Python 2.7 

sv.append(int(l[:2]))

What can i do?

scivision, 2017-10-20:
What error is it giving?
This program is automatically tested in Python 2.7/3.6
https://travis-ci.org/scivision/pyrinex

Cami23, 2017-10-24:
The error is in this codeline: sv.append(int(l[:2]))
This string contains a empty place, so it's impossible the cast to int. I'm trying to use a function to eliminate the empty place, however i dont know if it's functional

scivision, 2017-10-31:
OK this is a Python 2 specific problem I think, as in Python 3, int('      8      ') == 8

closed 2017-11-15 by scivision

Cami23, 2017-11-17:
You're right! 
It's compatible with Python3

scivision, 2017-11-17:
@Cami23 it should work in both Py 2.7 and 3 now