geospace-code/georinex #7: got an error when run  "python tests/test.py"

weisongwen, 2017-11-15:
Hi, @scivision  Thanks for your great work on this package.

I got an error when i run **python tests/test.py**, I have successfully catkin_make this package.

_EE
======================================================================
ERROR: test.test_obs
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/nose/case.py", line 197, in runTest
    self.test(*self.arg)
  File "/home/wenws/pyrinex/tests/test.py", line 15, in test_obs
    truth = read_hdf(str(rdir/'tests/demo.h5'),key='OBS')
  File "/usr/lib/python2.7/dist-packages/pandas/io/pytables.py", line 325, in read_hdf
    return f(store, True)
  File "/usr/lib/python2.7/dist-packages/pandas/io/pytables.py", line 317, in <lambda>
    key, auto_close=auto_close, **kwargs)
  File "/usr/lib/python2.7/dist-packages/pandas/io/pytables.py", line 643, in select
    raise KeyError('No object named %s in the file' % key)
KeyError: 'No object named OBS in the file'

======================================================================
ERROR: test.test_nav
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/nose/case.py", line 197, in runTest
    self.test(*self.arg)
  File "/home/wenws/pyrinex/tests/test.py", line 21, in test_nav
    truthnav = read_hdf(str(rdir/'tests/demo.h5'),key='NAV')
  File "/usr/lib/python2.7/dist-packages/pandas/io/pytables.py", line 323, in read_hdf
    store = HDFStore(path_or_buf, **kwargs)
  File "/usr/lib/python2.7/dist-packages/pandas/io/pytables.py", line 398, in __init__
    self.open(mode=mode, **kwargs)
  File "/usr/lib/python2.7/dist-packages/pandas/io/pytables.py", line 549, in open
    raise e
UnicodeDecodeError: 'utf8' codec can't decode byte 0xe0 in position 0: unexpected end of data

----------------------------------------------------------------------
Ran 2 tests in 0.020s

FAILED (errors=2)
Closing remaining open files:/home/wenws/pyrinex/tests/demo.h5...done_

scivision, 2017-11-15:
The program was not well compatible with Python 2.7. I made some changes to fix this. Please let me know how it's working for you with the current code.

weisongwen, 2017-11-16:
hi, @scivision 
I use "**pip install -e .**", it got an error as following:
_**pkg_resources.VersionConflict: (setuptools 3.3 (/usr/lib/python2.7/dist-packages), Requirement.parse('setuptools>=18.0'))**_

i tried to upgrade the setuptools by **pip install --upgrade setuptools**, but it says 
_**Installing collected packages: setuptools
  Found existing installation: setuptools 3.3
    Not uninstalling setuptools at /usr/lib/python2.7/dist-packages, owned by OS**_

do you have any idea about this?

scivision, 2017-11-16:
I would consider 

    pip install setuptools --user
    pip install -e .

weisongwen, 2017-11-17:
hi @scivision 
Thanks for your reply, I have successfully install this package. and can this package be used to calculate position of satellites?

best 
Welson

scivision, 2017-11-17:
Are you interested in using the Rinex Nav files to compute ECEF X,Y,Z position of satellites?

weisongwen, 2017-11-17:
hi @scivision  
Exactly, as I need to get pseudorange from .obs file and get satellites position from .nav file.

best 
Welson

scivision, 2018-03-21:
Do you have an example NAV file with satellite position?

weisongwen, 2018-03-22:
Hi, @scivision 
I do not have now, I did not focus on this currently,

best 
welson

charlesnuwan92, 2018-05-08:
Hello Scivision,

i got an error as bellow, do you know what i can do to make it resolve thank you !

KeyError                                  Traceback (most recent call last)
<ipython-input-36-9670afaf4efd> in <module>()
----> 1 obs=pr.rinexobs('GEOP107Q.18o')

~\AppData\Local\Continuum\anaconda3\lib\site-packages\pyrinex\__init__.py in rinexobs(fn, ofn)
    107         tic = time()
    108         lines = f.read().splitlines(True)
--> 109         header,version,headlines,headlength,obstimes,sats,svset = scan(lines)
    110         print(fn,'RINEX',version,'file',fn.stat().st_size//1000,'kB.')
    111         if fn.suffix=='.nc':

~\AppData\Local\Continuum\anaconda3\lib\site-packages\pyrinex\__init__.py in scan(L)
    154     header['APPROX POSITION XYZ'] = [float(j) for j in header['APPROX POSITION XYZ'].split()]
    155     #observation types
--> 156     header['# / TYPES OF OBSERV'] = header['# / TYPES OF OBSERV'].split()
    157     #turn into int number of observations
    158     header['# / TYPES OF OBSERV'][0] = int(header['# / TYPES OF OBSERV'][0])

KeyError: '# / TYPES OF OBSERV'

scivision, 2018-05-08:
@charlesnuwan92 also it looks like you may be running quite an old version of PyRinex, perhaps do a
```sh
git pull
``` 
or otherwise get the most recent code. Thanks

charlesnuwan92, 2018-05-09:
thank you for the reply, i updated the code to the latest one and try to run with you remo rinex files.and i got following error. By the way i am running this on jypterlab 

TypeError                                 Traceback (most recent call last)
<ipython-input-9-a3e10e99cac6> in <module>()
----> 1 obs,nav = pr.readrinex('demo3.10o')

~\AppData\Local\Continuum\anaconda3\lib\site-packages\pyrinex\__init__.py in readrinex(rinexfn, outfn, use, verbose)
     20         nav = rinexnav(rinexfn, outfn)
     21     elif fnl.endswith('o') or fnl.endswith('o.rnx'):
---> 22         obs = rinexobs(rinexfn, outfn, use=use, verbose=verbose)
     23     elif rinexfn.suffix.endswith('.nc'):
     24         nav = rinexnav(rinexfn)

~\AppData\Local\Continuum\anaconda3\lib\site-packages\pyrinex\__init__.py in rinexobs(fn, ofn, use, group, verbose)
     92     elif int(ver) == 3:
     93         if use is None or isinstance(use,str):
---> 94             obs = _scan3(fn, use, verbose)
     95         elif isinstance(use,(tuple,list,np.ndarray)):
     96             if len(use) == 1:

~\AppData\Local\Continuum\anaconda3\lib\site-packages\pyrinex\rinex3.py in _scan3(fn, use, verbose)
    160             darr = np.genfromtxt(BytesIO(raw.encode('ascii')), delimiter=(14,1,1)*Fmax)
    161 # %% Tselect one, few, or all satellites
--> 162             i = [i for i,s in enumerate(sv) if s[0] in use]
    163             garr = darr[i,:]
    164             gsv = np.array(sv)[i]

~\AppData\Local\Continuum\anaconda3\lib\site-packages\pyrinex\rinex3.py in <listcomp>(.0)
    160             darr = np.genfromtxt(BytesIO(raw.encode('ascii')), delimiter=(14,1,1)*Fmax)
    161 # %% Tselect one, few, or all satellites
--> 162             i = [i for i,s in enumerate(sv) if s[0] in use]
    163             garr = darr[i,:]
    164             gsv = np.array(sv)[i]

TypeError: argument of type 'NoneType' is not iterable

charlesnuwan92, 2018-05-09:
and i use  my file https://files.fm/u/95k952ck and get this error, really appreciate if you tell me whats wrong 

OverflowError                             Traceback (most recent call last)
<ipython-input-10-29955540fbcd> in <module>()
----> 1 obs,nav = pr.readrinex('GEOP107Q.18o')

~\AppData\Local\Continuum\anaconda3\lib\site-packages\pyrinex\__init__.py in readrinex(rinexfn, outfn, use, verbose)
     20         nav = rinexnav(rinexfn, outfn)
     21     elif fnl.endswith('o') or fnl.endswith('o.rnx'):
---> 22         obs = rinexobs(rinexfn, outfn, use=use, verbose=verbose)
     23     elif rinexfn.suffix.endswith('.nc'):
     24         nav = rinexnav(rinexfn)

~\AppData\Local\Continuum\anaconda3\lib\site-packages\pyrinex\__init__.py in rinexobs(fn, ofn, use, group, verbose)
     92     elif int(ver) == 3:
     93         if use is None or isinstance(use,str):
---> 94             obs = _scan3(fn, use, verbose)
     95         elif isinstance(use,(tuple,list,np.ndarray)):
     96             if len(use) == 1:

~\AppData\Local\Continuum\anaconda3\lib\site-packages\pyrinex\rinex3.py in _scan3(fn, use, verbose)
    144             time = datetime(int(l[2:6]), int(l[7:9]), int(l[10:12]),
    145                             hour=int(l[13:15]), minute=int(l[16:18]), second=int(l[19:21]),
--> 146                             microsecond=int(l[22:29])*1000000)
    147             if verbose:
    148                 print(time,'\r',end="")

OverflowError: Python int too large to convert to C long

scivision, 2018-05-09:
@charlesnuwan92  these two issues are fixed in the latest PyRinex v1.2.7
thanks for your patience.

closed 2018-05-09 by scivision

charlesnuwan92, 2018-05-10:
Hello Scivison,

Thank you for the correction. Your demo files run ok with the pyrinex but ;
i ran my file and looks like it does not record data to xarray my file can be download here https://files.fm/u/95k952ck
could you please observe the file and tell me whats wrong with it. 

<xarray.Dataset>
Dimensions:  (sv: 20, time: 933)
Coordinates:
  * sv       (sv) object 'C13' 'C27' 'C28' 'C30' 'G05' 'G14' 'G16' 'G20' ...
  * time     (time) datetime64[ns] 2018-04-17T16:19:25.999800 ...
Data variables:
    C1C      (time, sv) float64 nan nan nan nan nan nan nan nan nan nan nan ...
    C1Cssi   (time, sv) float64 nan nan nan nan nan nan nan nan nan nan nan ...
    L1C      (time, sv) float64 nan nan nan nan nan nan nan nan nan nan nan ...
    L1Clli   (time, sv) float64 nan nan nan nan nan nan nan nan nan nan nan ...
    L1Cssi   (time, sv) float64 nan nan nan nan nan nan nan nan nan nan nan ...
    D1C      (time, sv) float64 nan nan nan nan nan nan nan nan nan nan nan ...
    D1Cssi   (time, sv) float64 nan nan nan nan nan nan nan nan nan nan nan ...
    S1C      (time, sv) float64 nan nan nan nan nan nan nan nan nan nan nan ...
    S1Cssi   (time, sv) float64 nan nan nan nan nan nan nan nan nan nan nan ...
    C1B      (time, sv) float64 nan nan nan nan nan nan nan nan nan nan nan ...
    C1Bssi   (time, sv) float64 nan nan nan nan nan nan nan nan nan nan nan ...
    L1B      (time, sv) float64 nan nan nan nan nan nan nan nan nan nan nan ...
    L1Blli   (time, sv) float64 nan nan nan nan nan nan nan nan nan nan nan ...
    L1Bssi   (time, sv) float64 nan nan nan nan nan nan nan nan nan nan nan ...
    D1B      (time, sv) float64 nan nan nan nan nan nan nan nan nan nan nan ...
    D1Bssi   (time, sv) float64 nan nan nan nan nan nan nan nan nan nan nan ...
    S1B      (time, sv) float64 nan nan nan nan nan nan nan nan nan nan nan ...
    S1Bssi   (time, sv) float64 nan nan nan nan nan nan nan nan nan nan nan ...
    C2I      (time, sv) float64 nan nan nan nan nan nan nan nan nan nan nan ...
    C2Issi   (time, sv) float64 nan nan nan nan nan nan nan nan nan nan nan ...
    L2I      (time, sv) float64 nan nan nan nan nan nan nan nan nan nan nan ...
    L2Illi   (time, sv) float64 nan nan nan nan nan nan nan nan nan nan nan ...
    L2Issi   (time, sv) float64 nan nan nan nan nan nan nan nan nan nan nan ...
    D2I      (time, sv) float64 nan nan nan nan nan nan nan nan nan nan nan ...
    D2Issi   (time, sv) float64 nan nan nan nan nan nan nan nan nan nan nan ...
    S2I      (time, sv) float64 nan nan nan nan nan nan nan nan nan nan nan ...
    S2Issi   (time, sv) float64 nan nan nan nan nan nan nan nan nan nan nan ...
Attributes:
    filename:  GEOP107Q.18o

scivision, 2018-05-11:
Try the latest version `git pull`

I updated the Readme to note how to access particular satellite or data variables. NaN is a placeholder for particular satellites not seen at a given time step.

```python
Obs.sel(sv='G15')
```

charlesnuwan92, 2018-05-11:
Thank you !, everything works fine now .  for example If i want to plot time against pseudoranges/ signal strength/ Doppler  for all the constellations, how should i call the command.

charlesnuwan92, 2018-05-11:
Also when i input obs['C1B'] for example it only shows values of the first time step only ? it is possible to get all the time step values at once ?

thank you for the effort and really appreciate your help !

scivision, 2018-05-13:
From the RINEX 3.03 manual, `C1B` is for Galileo only. The data file you uploaded did not contain any Galileo data, it only contained GPS, GLONASS and Beidou data.

scivision, 2018-05-13:
A quick check would be 
```python
obs['C1B'].dropna(dim='time')
```
That returns
```
<xarray.DataArray 'C1B' (time: 0, sv: 20)>
```
which means that out of all 20 satellites observed throughout your whole data file, there was no `C1B` data.
I manually examined the data file to confirm this.

scivision, 2018-05-13:
Let's use the quantity `L1C` for an example, and pick the satellite `G14`, which we found in
```python
obs.sv
```

```python
l1c_g14 = obs['L1C'].sel(sv='G14').dropna(dim='time',how='all')
```
which returns
```
<xarray.DataArray 'L1C' (time: 927)>
```
and by inspection of this data vector and the text file, the data are extracted successfully.

This can be plotted:
```python
l1c_g14.plot()
```

charlesnuwan92, 2018-05-15:
Thank you Scivision this work perfectly now.

one last question to plot all the satellite's  pseudouranges against time at once do i need to do a for loop or is there a easy way.

really appreciate your effort and  time 
thank you,

scivision, 2018-05-15:
The `plotobs()` function does this. Simply:
```python
from matplotlib.pyplot import figure, show

ax = figure().gca()
ax.plot(obs.time, obs['L1C'])

show()

scivision, 2018-05-18:
What I would do is something like
```python
ax.plot(obs.time, obs['C1C'], label='C1C')
ax.legend()
```

charlesnuwan92, 2018-05-18:
Thanks for the reply.
Actually i tried that but instead of label 'C1C' i need for example the corresponding satellite maybe ''G14'' or ""E9"" so  tolabel the THE specific satellite of the constellation in the pseudorange graph. 

cheers

charlesnuwan92, 2018-05-18:
also s1b_e19 = obs1['S1B'].sel(sv='E19').dropna(dim='time',how='all'), in this code how do we select not only ''E19" but also another Galileo satellite.

thank you

scivision, 2018-05-23:
In the `plotobs()` function of plots.py, I show a looping over parameter plotting and that the legend is set (showing each SV name)
```python
for p in ('C1C','L1X'):
  ax = figure().gca()
  ax.plot(obs.time, obs[p])
  ax.legend(obs[p].sv.values.astype(str), loc='best')
```

scivision, 2018-05-23:
loading a single constellation of satellite data is *much* faster than loading all constellations. This would implicitly solve this issue.
```python
import pyrinex as pr
pr.readrinex('myfile.o',None,'E')
```
would load only Galileo data by the parameter `E`.  
ReadRinex.py allow this to be specified as the `-use` command line parameter.

If however you want to do this after loading all the data anyway, you can make a Boolean indexer
```python
Eind = obs.sv.to_index().str.startswith('E')

Edata = obs.isel(sv=Eind)  # any combination of other indices at same time or before/after also possible
```

charlesnuwan92, 2018-06-07:
Hi Michael.

thank you for the above 
I am trying to plot a graph for raw measurements. following is my code and i want to print the satellite id in the legend. I am having hard time to figure this out could you advice me how to do it

Satellite = "G14"
titles= ['Psedoranges of GPS and Glonass - '+ Satellite,'Carrier Phase', 'Doppler','Signal Strenght']
ylables= ['Pseudoranges','Phase','Doppler','signal strenght']
for sat,titles,ylables in zip(('C1C','L1C','D1C','S1C'),titles,ylables):
    Satobs=obsv[sat].sel(sv=Satellite,).dropna(dim='time',how='all')
    G25=obsv[sat].sel(sv='G25').dropna(dim='time',how='all')
    ax=figure().gca()
    Satobs.plot()
    plt.title(titles)
    plt.xlabel('Time,[s]')
    plt.ylabel(ylables)    
    **plt.legend(Satobs[:1],loc='center left', bbox_to_anchor=(1,0.5))**

scivision, 2018-06-07:
There were a few glitches in this plotting script. See
https://github.com/scivision/pyrinex/blob/master/PlotExample.py

charlesnuwan92, 2018-06-08:
Thank you for this  but i was trying to plot another graph using a different file path (different rinex file) and compare them , how do i call the coordinate sv ( for example 'G14') in the legend i tried this in the for loop and i get array not the SV id
plt.legend(Satobs[:1],loc='center left', bbox_to_anchor=(1,0.5))

charlesnuwan92, 2018-06-08:
3rd point is how to get time range i tried this
times = pd.date_range(start='2018-06-07T09:00:30',end='2018-06-07T09:15:00', freq='s')

then i tried with coords and i get error that coords is not define
Satobs2 = obsv[v].sel(sv=SV).dropna(coords = [time], dim='time', how='all')

and i tried other method i still get an error. couldnt find an example in Xarray

scivision, 2018-06-08:
I added this to `PlotExample.py`.  In general use boolean indexing for this type of operation.
```
i = (obs.time >= numpy.datetime64('2018-06-07T09:00:30')) & (obs.time <= '2018-06-07T09:15:00')

obs[v][i].sel(sv=SV) ...
...

when you want all the values (not limiting in time), let `i = slice(None)`

charlesnuwan92, 2018-06-11:
Thank you for this Michael!
just wondering whether i can combine two high rate Rinex files ( 15 min obs each ) together for analysis using your library

scivision, 2018-06-12:
I made some notes on this in https://github.com/scivision/pyrinex/blob/master/README.md#analysis

You want to load/save the data from each file as NetCDF with the `-o filename.nc` option.
Then concat/merge the data and save with `obs.to_netcdf('joined.nc', group='OBS')`

charlesnuwan92, 2018-06-18:
Thank you Michael,

I was trying to read a navigation file using your library and i get an error saying "could not broadcast input array from shape (28) into shape (27)"
i used a  GNSS mixed Nav rinex 3 file as follow:
ftp://cddis.gsfc.nasa.gov/pub/gps/data/daily/2018/158/18p/VEN100ITA_R_20181580000_01D_MN.rnx.gz

i really appreciate If you could tell me whats the issue here

scivision, 2018-06-21:
This is fixed by v1.4.0 Pyrinex

charlesnuwan92, 2018-06-21:
HI Michael thank you for this. really appreciate your effort
i have 2 question 
1. when i run NAV rinex it reads fine but i get a warning as such: 

_WARNING:root:duplicate times detected, skipping SV E01
WARNING:root:duplicate times detected, skipping SV E02
WARNING:root:duplicate times detected, skipping SV E03
WARNING:root:duplicate times detected, skipping SV E04
WARNING:root:duplicate times detected, skipping SV E05
WARNING:root:duplicate times detected, skipping SV E07
WARNING:root:duplicate times detected, skipping SV E08
WARNING:root:duplicate times detected, skipping SV E09
WARNING:root:duplicate times detected, skipping SV E11
WARNING:root:duplicate times detected, skipping SV E12
WARNING:root:duplicate times detected, skipping SV E14
WARNING:root:duplicate times detected, skipping SV E18
WARNING:root:duplicate times detected, skipping SV E19
WARNING:root:duplicate times detected, skipping SV E24
WARNING:root:duplicate times detected, skipping SV E25
WARNING:root:duplicate times detected, skipping SV E26
WARNING:root:duplicate times detected, skipping SV E30_

and there is not nav data stored in the Xarray data frame for Galileo.

2/ when i  convert the Xarray data frame to pandas data frame how do i get only SV column (or select sv column ). I used multi indexing selection then i get an error saying key error 'sv'

scivision, 2018-06-21:
This is fixed by keeping the first time instance instead of discarding. From what I've seen, when the time is duplicated, all the data at that time is duplicated too.

Indexing the Dataframe that results from `df = rinex.to_dataframe()` may be indexed by SV by:
`df.loc['G12'].dropna(0,'all')`

charlesnuwan92, 2018-06-28:
Thank you Michael,

From what i see that the readnav does not record any Galileo satellites and it produces wrong UTC time for all the satellites. I observe this by comparing the navigation file and the output from your pyrinex and also by calling the Nav.coords["time"] , Nav.sv (no Galileo satellite)

I uploaded my nav file with a image of the output - https://files.fm/u/yfedpd5b

I believe this is a small issue and thank you for helping me out !

scivision, 2018-07-30:
Hello @charlesnuwan92 This is fixed by https://github.com/scivision/georinex/releases v1.5.3