geospace-code/georinex #8: how can this package be changed to read the rinex3.02 .nav file

weisongwen, 2017-11-17:
Hi, @scivision , thanks for your work in this package.
Thanks for your work here. how can this package be changed to read the rinex3.02 .nav file? my .nav file is like this (rinex 3.2):
``` 
3.02           N: GNSS NAV DATA    G: GPS              RINEX VERSION / TYPE
LgsRinex v1.7.7.0   SMO                 20171117 051209 UTC PGM / RUN BY / DATE
GPSA   1.1176D-08 -1.4901D-08 -5.9605D-08  1.1921D-07       IONOSPHERIC CORR
GPSB   9.8304D+04 -1.1469D+05 -1.9661D+05  7.2090D+05       IONOSPHERIC CORR
GPUT -3.7252902985D-09-1.065814104D-14  61440 1976          TIME SYSTEM CORR
    18    18  1929     7                                    LEAP SECONDS
                                                            END OF HEADER
G01 2017 11 17 00 00 00-1.082336530089D-05-2.387423592154D-12 0.000000000000D+00
     2.600000000000D+01 4.312500000000D+01 4.455542601534D-09 2.261013945883D+00
     2.192333340645D-06 7.114754174836D-03 7.407739758492D-06 5.153672552109D+03
     4.320000000000D+05-9.126961231232D-08 7.577775093435D-01-7.450580596924D-08
     9.691207199508D-01 2.407187500000D+02 6.318014759246D-01-7.981403982171D-09
     5.718095441587D-10 1.000000000000D+00 1.975000000000D+03 0.000000000000D+00
     2.000000000000D+00 0.000000000000D+00 5.587935447693D-09 2.600000000000D+01
     4.247400000000D+05                   
G02 2017 11 17 04 00 00 2.789050340652D-04-9.663381206337D-12 0.000000000000D+00
     6.100000000000D+01 4.615625000000D+01 4.950563514683D-09-1.495754475149D+00
     2.572312951088D-06 1.728782569990D-02 7.607042789459D-06 5.153617258072D+03
     4.464000000000D+05 4.153698682785D-07 7.005942974446D-01 3.054738044739D-07
     9.481762359175D-01 2.264062500000D+02-1.938649816409D+00-8.148910879413D-09
     4.689481025721D-10 1.000000000000D+00 1.975000000000D+03 0.000000000000D+00
     2.000000000000D+00 0.000000000000D+00-2.048909664154D-08 6.100000000000D+01
     4.391400000000D+05                   
G03 2017 11 17 00 00 00-4.102196544409D-05 5.570655048359D-12 0.000000000000D+00
     6.200000000000D+01-3.440625000000D+01 4.867345637649D-09 1.199389953372D+00
    -1.976266503334D-06 1.129540265538D-03 5.628913640976D-06 5.153610057831D+03
     4.320000000000D+05 1.117587089539D-08 1.800773457900D+00 3.911554813385D-08
     9.598706640660D-01 2.691562500000D+02 5.853639356718D-01-8.238200344124D-09
    -3.446572194310D-10 1.000000000000D+00 1.975000000000D+03 0.000000000000D+00
     2.000000000000D+00 0.000000000000D+00 2.328306436539D-09 6.200000000000D+01
     4.247400000000D+05                   
G05 2017 11 17 04 00 00-2.250215038657D-05 1.136868377216D-12 0.000000000000D+00
     1.010000000000D+02-3.156250000000D+01 5.160929461567D-09 1.019456906164D+00
    -1.467764377594D-06 5.289388587698D-03 5.319714546204D-06 5.153583730698D+03
     4.464000000000D+05-7.823109626770D-08 1.781452415021D+00 4.842877388000D-08
     9.467839021727D-01 2.704687500000D+02 6.257362175680D-01-8.413207908120D-09
    -3.964450712157D-10 1.000000000000D+00 1.975000000000D+03 0.000000000000D+00
     2.000000000000D+00 0.000000000000D+00-1.071020960808D-08 1.010000000000D+02
     4.391400000000D+05                   
```

weisongwen, 2017-11-20:
Hi, @scivision  
I finally find got the 2.1 version of .nav file. 
Thanks 

Best 
Welson

scivision, 2017-11-20:
@weisongwen I implemented RINEX3 NAV for SBAS only. Will look at GPS next.

scivision, 2017-11-20:
@weisongwen let me know if it doesn't work for you on RINEX3 NAV

closed 2017-11-20 by scivision