geospace-code/georinex #9: The time for gps.getSatXYZ(nav,1,arr) function

weisongwen, 2017-11-21:
Hi @scivision  Thanks for your effort in this package.
when I use following function, I found that the output satellite position is not right, 
```
satpos1 = gps.getSatXYZ(nav,1,arr)
satpos2 = gps.getSatXYZ(nav,2,arr)
satpos3 = gps.getSatXYZ(nav,3,arr)
```
the parameter **arr** in function **gps.getSatXYZ** is the UTC time from .obs file? I use the following code to get the time for function **gps.getSatXYZ**.
```
arr_test = np.array([gps_receiver_time + datetime.timedelta(minutes=i) for i in range(2)])#make sure the length of arr_test is not 0

```
**gps_receiver_time** is the UTC time provided by .obs file.

and then compute satellite 3 position:
```
satellite_position=gps.getSatXYZ(nav,3,arr_test)
```
I not sure whether the 
Best 
Welson

scivision, 2017-11-22:
I don't think there's a `getSatXYZ` in this package?

weisongwen, 2017-11-27:
Hi， @scivision  sorry for this mistake.
the function```satellite_position=gps.getSatXYZ(nav,3,arr_test)``` is acquired from another package, sorry for this .

Best 
Welson

closed 2017-11-28 by scivision