geospace-code/georinex #14: ValueError running pyrinex

ianmarang, 2018-07-02:
Hi

I'm getting an error when first reading in an obs file.  

import pyrinex as pr
obs,nav = pr.readrinex('../Sample/14601736.18o')

Full StackTrace:

---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-4-bc870e0505e0> in <module>()
      2 srcDir = '/Users/ianmarang/Documents/Project_docs/Crop/Narrabri_farm_data/DGPS_SNR/Sample/'
      3 
----> 4 obs,nav = pr.readrinex(srcDir+'14601736.18o')
      5 print(obs)
      6 print(nav)

~/anaconda2/envs/py3/lib/python3.6/site-packages/pyrinex/__init__.py in readrinex(rinexfn, outfn, use, verbose)
     29         nav = rinexnav(rinexfn, outfn)
     30     elif fnl.endswith('o') or fnl.endswith('o.rnx'):
---> 31         obs = rinexobs(rinexfn, outfn, use=use, verbose=verbose)
     32     elif rinexfn.suffix.endswith('.nc'):
     33         nav = rinexnav(rinexfn)

~/anaconda2/envs/py3/lib/python3.6/site-packages/pyrinex/__init__.py in rinexobs(fn, ofn, use, group, verbose)
     97     ver = getRinexVersion(fn)
     98     if int(ver) == 2:
---> 99         obs = _scan2(fn, use, verbose)
    100     elif int(ver) == 3:
    101         obs = _scan3(fn, use, verbose)

~/anaconda2/envs/py3/lib/python3.6/site-packages/pyrinex/rinex2.py in _scan2(fn, use, verbose)
    162                 break
    163 
--> 164             eflag = int(ln[28])
    165             if eflag not in (0, 1, 5, 6):  # EPOCH FLAG
    166                 print(eflag)

ValueError: invalid literal for int() with base 10: 'i'

Conda details:
active environment : py3
    active env location : /Users/ianmarang/anaconda2/envs/py3
            shell level : 1
       user config file : /Users/ianmarang/.condarc
 populated config files :
          conda version : 4.5.4
    conda-build version : not installed
         python version : 2.7.14.final.0
       base environment : /Users/ianmarang/anaconda2  (writable)
           channel URLs : https://repo.anaconda.com/pkgs/main/osx-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/free/osx-64
                          https://repo.anaconda.com/pkgs/free/noarch
                          https://repo.anaconda.com/pkgs/r/osx-64
                          https://repo.anaconda.com/pkgs/r/noarch
                          https://repo.anaconda.com/pkgs/pro/osx-64
                          https://repo.anaconda.com/pkgs/pro/noarch
          package cache : /Users/ianmarang/anaconda2/pkgs
                          /Users/ianmarang/.conda/pkgs
       envs directories : /Users/ianmarang/anaconda2/envs
                          /Users/ianmarang/.conda/envs
               platform : osx-64
             user-agent : conda/4.5.4 requests/2.14.2 CPython/2.7.14 Darwin/14.5.0 OSX/10.10.5
                UID:GID : 505:20
             netrc file : None
           offline mode : False

# Name                    Version                   Build  Channel
appnope                   0.1.0            py36hf537a9a_0
backcall                  0.1.0                    py36_0
blas                      1.0                         mkl
bzip2                     1.0.6                h1de35cc_5
ca-certificates           2018.03.07                    0
cairo                     1.14.12              hc4e6be7_3
certifi                   2018.4.16                py36_0
cftime                    1.0.0                     <pip>
curl                      7.60.0               ha441bb4_0
Cython                    0.28.3                    <pip>
decorator                 4.3.0                    py36_0
expat                     2.2.5                hb8e80ba_0
fontconfig                2.13.0               h5d5b041_1
freetype                  2.9.1                hb4e5f40_0
freexl                    1.0.5                h1de35cc_0
gdal                      2.2.4            py36h1702cab_1
geos                      3.6.2                h5470d99_2
gettext                   0.19.8.1             h15daf44_3
giflib                    5.1.4                h1035379_1
glib                      2.56.1               h35bc53a_0
hdf4                      4.2.13               h39711bb_2
hdf5                      1.8.18               h017327d_1
icu                       58.2                 h4b95b61_1
intel-openmp              2018.0.3                      0
ipykernel                 4.8.2                    py36_0
ipython                   6.4.0                    py36_0
ipython_genutils          0.2.0            py36h241746c_0
jedi                      0.12.0                   py36_1
jpeg                      9b                   he5867d9_2
json-c                    0.13.1               h3efe00b_0
jupyter_client            5.2.3                    py36_0
jupyter_core              4.4.0            py36h79cf704_0
kealib                    1.4.7                haa18932_5
libboost                  1.65.1               hcc95346_4
libcurl                   7.60.0               hf30b1f0_0
libcxx                    4.0.1                h579ed51_0
libcxxabi                 4.0.1                hebd6815_0
libdap4                   3.19.0               hefe82ba_2
libedit                   3.1.20170329         hb402a30_2
libffi                    3.2.1                h475c297_4
libgdal                   2.2.4                h2fbb439_1
libgfortran               3.0.1                h93005f0_2
libiconv                  1.15                 hdd342a3_7
libkml                    1.3.0                hc0cae1e_3
libnetcdf                 4.4.1.1              h1d6afd8_8
libpng                    1.6.34               he12f830_0
libpq                     10.3                 hf30b1f0_0
libsodium                 1.0.16               h3efe00b_0
libspatialite             4.3.0a              h0a2fa02_18
libssh2                   1.8.0                h322a93b_4
libtiff                   4.0.9                hcb84e12_1
libuuid                   1.0.3                h6bb4b03_2
libxml2                   2.9.8                hab757c2_1
mkl                       2018.0.3                      1
mkl_fft                   1.0.1            py36h917ab60_0
mkl_random                1.0.1            py36h78cc56f_0
ncurses                   6.1                  h0a44026_0
netCDF4                   1.4.0                     <pip>
numpy                     1.14.5           py36h9bb19eb_2
numpy-base                1.14.5           py36ha9ae307_2
openjpeg                  2.3.0                hb95cd4c_1
openssl                   1.0.2o               h26aff7b_0
pandas                    0.23.1           py36h1702cab_0
parso                     0.2.1                    py36_0
pcre                      8.42                 h378b8a2_0
pexpect                   4.6.0                    py36_0
pickleshare               0.7.4            py36hf512f8e_0
pip                       10.0.1                   py36_0
pixman                    0.34.0               hca0a616_3
poppler                   0.65.0               ha097c24_1
poppler-data              0.4.9                         0
proj4                     4.9.3                h3f1bf9d_7
prompt_toolkit            1.0.15           py36haeda067_0
ptyprocess                0.6.0                    py36_0
pygments                  2.2.0            py36h240cd3f_0
pyrinex                   1.4.1                     <pip>
python                    3.6.6                hc167b69_0
python-dateutil           2.7.3                    py36_0
pytz                      2018.4                   py36_0
pyzmq                     17.0.0           py36h1de35cc_1
readline                  7.0                  hc1231fa_4
setuptools                39.2.0                   py36_0
simplegeneric             0.8.1                    py36_2
six                       1.11.0           py36h0e22d5e_1
sqlite                    3.24.0               ha441bb4_0
tk                        8.6.7                h35a86e2_3
tornado                   5.0.2                    py36_0
traitlets                 4.3.2            py36h65bd3ce_0
wcwidth                   0.1.7            py36h8c6ec74_0
wheel                     0.31.1                   py36_0
xarray                    0.10.7                    <pip>
xerces-c                  3.2.1                h44e365a_0
xz                        5.2.4                h1de35cc_4
zeromq                    4.2.5                h378b8a2_0
zlib                      1.2.11               hf3cbc9b_2

Looks like its failing when trying to confirm the version of the rinex obs file, but I've never used this before today so I could use some help.

Cheers
Ian

scivision, 2018-07-26:
This looks like a possible malformed RINEX file, can you upload/send/link this file?

ianmarang, 2018-07-26:
Hi Michael,

Thanks for getting back to me.  The file is attached.

Regards

Ian Marang
PhD Student
School of Environment and Life Sciences
The University of Sydney

m: 0432941663

scivision, 2018-07-26:
Hi @ianmarang the email reply went to GitHub, which stripped the file off. Are you able to upload it say to Google Drive or Dropbox etc. to share?

ianmarang, 2018-07-30:
Hi Michael,

Sure:

https://drive.google.com/drive/folders/1sMEh_iLBV726FV5DLyUO-F44KsJwtCc8?usp=sharing

Thanks

Ian Marang
PhD Student
School of Environment and Life Sciences
The University of Sydney

m: 0432941663

ianmarang, 2018-07-30:
I should note that it’s only the 14601736.18o file that we reference in the pyrinex call.

Regards

Ian Marang
PhD Student
School of Environment and Life Sciences
The University of Sydney

m: 0432941663

https://drive.google.com/drive/folders/1sMEh_iLBV726FV5DLyUO-F44KsJwtCc8?usp=sharing

scivision, 2018-07-30:
This is fixed in version 1.5.2, it was the mangling of the file from the conversion program.

closed 2018-07-30 by scivision