geospace-code/georinex #16: 'IndexError: too many indices for array' for large RINEX 3 files

VOMAY, 2018-07-13:
Dear scivision,

I spend the day checking out pyrinex and am starting to like it. However I am having issue reading large RINEX3 files (24h@30s). In this case it crashed at 01:26:30.

I work at ESA's Navigation Support Office and we do a lot of RINEX analysis, mostly of 24h files. I see great potential in this tool for our daily work.

I am looking forward to hear from you.

``` python
a= pr.readrinex('ABMF00GLP_R_20181330000_01D_30S_MO.rnx')
Traceback (most recent call last):02:30 2018-05-13 00:04:30   2018-05-13 00:09:00       2018-05-13 00:16:30  2018-05-13 00:19:30 2018-05-13 00:21:30  2018-05-13 00:24:30 2018-05-13 00:26:00 2018-05-13 00:27:30   2018-05-13 00:31:30 2018-05-13 00:33:30              2018-05-13 00:47:30    2018-05-13 00:52:00           2018-05-13 01:06:00  2018-05-13 01:09:00 2018-05-13 01:10:30  2018-05-13 01:13:30 2018-05-13 01:15:00      2018-05-13 01:21:00    2018-05-13 01:25:00 2018-05-13 01:26:30   

  File "<ipython-input-87-b3eefa642c66>", line 1, in <module>
    a= pr.readrinex('C:/Users/xxx/Desktop/ABMF00GLP_R_20181330000_01D_30S_MO.rnx')

  File "C:\Users\xxx\AppData\Local\Continuum\anaconda3\lib\site-packages\pyrinex\__init__.py", line 31, in readrinex
    obs = rinexobs(rinexfn, outfn, use=use, verbose=verbose)

  File "C:\Users\xxx\AppData\Local\Continuum\anaconda3\lib\site-packages\pyrinex\__init__.py", line 101, in rinexobs
    obs = _scan3(fn, use, verbose)

  File "C:\Users\xxx\AppData\Local\Continuum\anaconda3\lib\site-packages\pyrinex\rinex3.py", line 258, in _scan3
    garr = darr[si, :]

IndexError: too many indices for array

```
I am running pyrinex 1.4.1. find the other versions below:
<details>
# Name                    Version                   Build  Channel
_ipyw_jlab_nb_ext_conf    0.1.0            py36he6757f0_0
alabaster                 0.7.10           py36hcd07829_0
anaconda                  5.2.0                    py36_3
anaconda-client           1.6.14                   py36_0
anaconda-navigator        1.8.7                    py36_0
anaconda-project          0.8.2            py36hfad2e28_0
asn1crypto                0.24.0                   py36_0
astroid                   1.6.3                    py36_0
astropy                   3.0.2            py36h452e1ab_1
attrs                     18.1.0                   py36_0
babel                     2.5.3                    py36_0
backcall                  0.1.0                    py36_0
backports                 1.0              py36h81696a8_1
backports.shutil_get_terminal_size 1.0.0            py36h79ab834_2
beautifulsoup4            4.6.0            py36hd4cc5e8_1
bitarray                  0.8.1            py36hfa6e2cd_1
bkcharts                  0.2              py36h7e685f7_0
blas                      1.0                         mkl
blaze                     0.11.3           py36h8a29ca5_0
bleach                    2.1.3                    py36_0
blosc                     1.14.3               he51fdeb_0
bokeh                     0.12.16                  py36_0
boto                      2.48.0           py36h1a776d2_1
bottleneck                1.2.1            py36hd119dfa_0
bzip2                     1.0.6                hfa6e2cd_5
ca-certificates           2018.03.07                    0
certifi                   2018.4.16                py36_0
cffi                      1.11.5           py36h945400d_0
cftime                    1.0.0b1          py36h452e1ab_0
chardet                   3.0.4            py36h420ce6e_1
click                     6.7              py36hec8c647_0
cloudpickle               0.5.3                    py36_0
clyent                    1.2.2            py36hb10d595_1
colorama                  0.3.9            py36h029ae33_0
comtypes                  1.1.4                    py36_0
conda                     4.5.8                    py36_0
conda-build               3.10.5                   py36_0
conda-env                 2.6.0                h36134e3_1
conda-verify              2.0.0            py36h065de53_0
console_shortcut          0.1.1                h6bb2dd7_3
[... 184 lines cut ...]
typing                    3.6.4                    py36_0
unicodecsv                0.14.1           py36h6450c06_0
urllib3                   1.22             py36h276f60a_0
vc                        14                   h0510ff6_3
vs2015_runtime            14.0.25123                    3
wcwidth                   0.1.7            py36h3d5aa90_0
webencodings              0.5.1            py36h67c50ae_1
werkzeug                  0.14.1                   py36_0
wheel                     0.31.1                   py36_0
widgetsnbextension        3.2.1                    py36_0
win_inet_pton             1.0.1            py36he67d7fd_1
win_unicode_console       0.5              py36hcdbd4b5_0
wincertstore              0.2              py36h7fe50ca_0
winpty                    0.4.3                         4
wrapt                     1.10.11          py36he5f5981_0
xarray                    0.10.7                    <pip>
xlrd                      1.1.0            py36h1cb58dc_1
xlsxwriter                1.0.4                    py36_0
xlwings                   0.11.8                   py36_0
xlwt                      1.3.0            py36h1a4751e_0
yaml                      0.1.7                hc54c509_2
zeromq                    4.2.5                hc6251cf_0
zict                      0.1.3            py36h2d8e73e_0
zlib                      1.2.11               h8395fce_2
</details>

VOMAY, 2018-07-16:
Dear scivision,

I looked into this a bit in depth and the figured out that the crash is indeed unrelated to the file size. In fact, its caused by the individual irregularities in the RINEX files I tested. Strange things can happen in Receivers and end up in the RINEX.

[ABMF00GLP_R_20181330000_01D_30S_MO.zip](https://github.com/scivision/pyrinex/files/2197760/ABMF00GLP_R_20181330000_01D_30S_MO.zip)

Anyway, in case you are planning to continue developing this handy piece of software, I am happy to give you some input. Just get in touch.

Best Regards
Volker

scivision, 2018-07-25:
Thanks Volker @VOMAY for your patience, I am looking at this now.

VOMAY, 2018-07-26:
Thanks!
I will test this mid August.