geospace-code/georinex #21: Installation error/ Rinex nav read error for mixed Rinex files

charlesnuwan92, 2018-07-31:
Hi Michael,

when i was installing Georinex i got this error message but the package installed fine.
 

Requirement already satisfied: pytz>=2011k in c:\users\gssc\appdata\local\continuum\anaconda3\lib\site-packages (from pandas>=0.19.2->xarray->georinex)
Building wheels for collected packages: georinex
  Running setup.py bdist_wheel for georinex ... error
  Complete output from command C:\Users\GSSC\AppData\Local\Continuum\anaconda3\python.exe -u -c "import setuptools, tokenize;__file__='C:\\Users\\GSSC\\AppData\\Local\\Temp\\pip-build-8bww53q8\\georinex\\setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d C:\Users\GSSC\AppData\Local\Temp\tmptm1u2xrhpip-wheel- --python-tag cp36:
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build\lib
  creating build\lib\georinex
  copying georinex\io.py -> build\lib\georinex
  copying georinex\keplerian.py -> build\lib\georinex
  copying georinex\nav2.py -> build\lib\georinex
  copying georinex\nav3.py -> build\lib\georinex
  copying georinex\obs2.py -> build\lib\georinex
  copying georinex\obs3.py -> build\lib\georinex
  copying georinex\plots.py -> build\lib\georinex
  copying georinex\__init__.py -> build\lib\georinex
  running egg_info
  writing georinex.egg-info\PKG-INFO
  writing dependency_links to georinex.egg-info\dependency_links.txt
  writing entry points to georinex.egg-info\entry_points.txt
  writing requirements to georinex.egg-info\requires.txt
  writing top-level names to georinex.egg-info\top_level.txt
  reading manifest file 'georinex.egg-info\SOURCES.txt'
  writing manifest file 'georinex.egg-info\SOURCES.txt'
  installing to build\bdist.win-amd64\wheel
  running install
  running install_lib
  creating build\bdist.win-amd64
  creating build\bdist.win-amd64\wheel
  creating build\bdist.win-amd64\wheel\georinex
  copying build\lib\georinex\io.py -> build\bdist.win-amd64\wheel\.\georinex
  copying build\lib\georinex\keplerian.py -> build\bdist.win-amd64\wheel\.\georinex
  copying build\lib\georinex\nav2.py -> build\bdist.win-amd64\wheel\.\georinex
  copying build\lib\georinex\nav3.py -> build\bdist.win-amd64\wheel\.\georinex
  copying build\lib\georinex\obs2.py -> build\bdist.win-amd64\wheel\.\georinex
  copying build\lib\georinex\obs3.py -> build\bdist.win-amd64\wheel\.\georinex
  copying build\lib\georinex\plots.py -> build\bdist.win-amd64\wheel\.\georinex
  copying build\lib\georinex\__init__.py -> build\bdist.win-amd64\wheel\.\georinex
  running install_egg_info
  Copying georinex.egg-info to build\bdist.win-amd64\wheel\.\georinex-1.5.3-py3.6.egg-info
  running install_scripts
  error: [Errno 2] No such file or directory: 'LICENSE'

  ----------------------------------------
  Failed building wheel for georinex
  Running setup.py clean for georinex
Failed to build georinex
Installing collected packages: georinex
  Running setup.py install for georinex ... done
Successfully installed georinex-1.5.3

Then i read the mixed Rinex file using the updated version and i examined each data point visually by exporting the data to a excel file. I see (specially Galileo sats) the dates are wrong and some attribute are empty ( in Rinex file they are not). I don't know whether this has something to do with errors during the installation. 
Really appreciate the effort you put to make this package.
I hope you will be able to correct this.

Rinex file, excel file -- https://files.fm/u/cu7wrt3z

Cheers
Best regards
Charles

scivision, 2018-08-02:
On the latest "master" branch on GitHub here, I enabled the "use=" option for NAV3 files as well, I would try this and see if you get what you expect e.g. 
```sh
ReadRinex tests/VILL00ESP_R_20181700000_01D_MN.rnx.gz  -u E 
```

or 
```python
import georinex as gr

dat = gr.readrinex('tests/VILL00ESP_R_20181700000_01D_MN.rnx.gz', use='E')
```

scivision, 2018-08-02:
The "use" option for OBS and NAV files speeds the reading and reduces extraneous fields by only reading one system type e.g. Galileo, GPS ,etc.

scivision, 2018-08-13:
@charlesnuwan92 let me know if this is still an issue. Thanks.

closed 2018-08-13 by scivision