geospace-code/georinex #22: Proposal: Add use option also for three character observation type

VOMAY, 2018-08-06:
Dear Scivision,

I can confirm that your effort to speed up the parsing, by making the indicators optional, was successful.

Unfortunately, the parsing is still a bit slow. I have another proposal to speed it up:

The **use** argument is very useful to speed up the parsing, by selecting only the constellations/satellites you want. Especially ignoring the GEOs makes a big different, since they are recorded every single epoch. How do you feel about adding an option to allow a similar selection for the signal types? 

Here is an example of what I would like to do:

``` python
#Only read Galileo E1 Code
gr.readrinex('test.rnx', use='E', signal='C1C')

#Read all code and phase observations of all constellations but GEOs
gr.readrinex('test.rnx', use=['E', 'G', 'C', 'R', 'J'], signal=['C', 'L'] )
```
I am convinced the potential time saving is substantial. 

Implementing this, one may consider changing the **use** keyword, since it is not unambiguous anymore. But this is a detail.

Thanks a lot for your effort. Keep up the good work!

Best Regards
Volker

scivision, 2018-08-06:
Thanks Volker, yes due to read buffering and the like, it is very speedy to just skip satellites. 
Right, for the command-line `ReadRinex` as well as the direct Python API, a new keyword is needed to include specific measurements.
Yes I think this will at least offer substantial memory benefits, as often we use only a few of the ~30 measurements available for each satellite

VOMAY, 2018-08-06:
This works, but it is not yet perfectly user friendly. I would prefer wildcards, so I do not have to select the whole list of signals.

Look at the example I posted:
``` python
#Read all code and phase observations of all constellations but GEOs
gr.readrinex('test.rnx', use=['E', 'G', 'C', 'R', 'J'], signal=['C', 'L'] )
```
I would like to select all available Code and/or Phase observations with a simple command. This requires to check the length of the **meas** input (1, 2 or 3 characters) and compare those to the first characters of the signal names (Regex might be helpful here). This would simplify the use a lot. Alternatively you could also just discriminate between C, L, D and S. The RINEX files I am analyzing often have more than 20 different signals. I'd like to AVOID something like this:

``` python
#Read all code and phase observations of all constellations but GEOs (your current implementation) (the signal list is still not complete)
gr.readrinex('test.rnx', use=['E', 'G', 'C', 'R', 'J'], signal=['C1C', 'C1W', 'C2W' 'C2S', 'C2L', 'C2P', 'C5I', 'C5Q', 'C5X', 'C7I', 'C7Q', 'C7X' 'C8Q', 'L1C', 'C1C', 'L2S', 'L2L', 'C2P', 'L5I', 'L5Q', 'L5X', 'L7I', 'L7Q', 'L7X' 'L8Q'] )
```

The same could be done for the satellites (**use**), e.g.: 'E0?' for E01-E09

scivision, 2018-08-06:
Note I found a bug--it's actually not indexing properly with this new "meas=" option. Fix on the way soon.

scivision, 2018-08-06:
Note: https://github.com/scivision/georinex/commit/bf068c32392a19d6e0802364b34f74ceb4f15112 fixes a previous error in the new `meas=` feature you were testing. 
 
I agree, auto-wildcard based on simple regex is highly useful as well

VOMAY, 2018-08-06:
Thanks for focusing on this issue.

On my side, I am in Germany and will call it day now, to get home before sunset. Talk to you tomorrow.

scivision, 2018-08-08:
This is implemented for `meas=` by checking the first character(s).
`meas='C'` or `meas=['C']` takes all measurements starting with `C` like C1, C2`
`meas=`C1` takes all measurements starting with `C1` and so on.

VOMAY, 2018-08-08:
I just tested it for OBS3 and it works like a charm. Thanks a lot! This will save me a lot of time.

closed 2022-09-19 by VOMAY