geospace-code/georinex #24: Proposal: Convert Space to 0 in PRN

VOMAY, 2018-08-07:
Dear Scivision

some RINEX writers have the habit to write the 1-digit PRN numbers without the leading 0, e.g. : 

``` 
17  4 29  0  2  0.0000000  0 23R 2R 8G19R18R 1G14R 9R19G 6G22G32G 2
```
instead of
``` 
17  4 29  0  2  0.0000000  0 23R02R08G19R18R01G14R09R19G06G22G32G02
```

I have seen this in NAV and OBS files, in both RINEX 2 and 3. In general, this is not an issue for georinex. However, the space ends up in the **sv** coordinate ticks. This makes it hard to compare the same satellite in two RINEX files, recorded by different receivers.

I propose to check the second character of each PRN for a space and replace it by a 0.

Best Regards
Volker

VOMAY, 2018-08-10:
Dear scivision,

I think the way you implemented this fix, is not optimal. You are checking for the spaces already in the read loop, therefore you are running the following loop 2880 times for a 24h@30s file. Alternatively you could just do this once on the final `sv` list, before you allocate the `xarray.DataSet`. Your approach is better in case a RINEX files has both `E01` and `E 1`, but I have never seen this. A RinexWriter does either one or the other.

``` python
for i, ln in zip(range(Nsv), f):
sv.append(ln[:3].replace(' ', '0'))
raw += ln[3:]
```

scivision, 2018-08-13:
Fixed in https://github.com/scivision/georinex/releases/tag/v1.6.0.1