geospace-code/georinex #25: rinexheader does not parse rinex file with more than 9 observation types

serioca, 2018-08-13:
submitting to rinexheader  function a rinex file with more than 9 observation types, like the following:
=========================================================
   10    C1    L1    S1    P2    L2    S2    C2    C5    L5# / TYPES OF OBSERV
           S5                                                                # / TYPES OF OBSERV
=========================================================
I got a parsing error and I fixed It modifing module obs2.py as shown below:

BEFORE FIXING
137        if '# / TYPES OF OBSERV' in h:
138           if Nobs is None:
139                Nobs = int(c[:6])
140                c = c[6:].split()

AFTER FIXING
137        if '# / TYPES OF OBSERV' in h:
138           if Nobs is None:
139                Nobs = int(c[:6])
140           c = c[6:].split()

Regards

Sergio Bruni

serioca, 2018-08-13:
Sorry, my issue was not clear. To fix the problem I put the line number 140 out of  the "if" of line number 138.
Regards
Sergio Bruni

scivision, 2018-08-13:
fixed https://github.com/scivision/georinex/releases/tag/v1.6.0.2   thanks!

scivision, 2018-08-13:
Note: I happened to look at your fork.  A lot has been added since then, including an OBS2 header reader: https://github.com/scivision/georinex/blob/master/georinex/obs2.py#L140  `obsheader2()`

closed 2018-08-13 by scivision