geospace-code/georinex #26: np.genfromtxt time consumption

VOMAY, 2018-08-24:
Dear Scivision,

I put your latest release v1.6.1 in a profiler. After getting rid of xarrax.concat and .merge, the new bottle neck is the 'np.genfromtxt' function:
``` python
darr = np.genfromtxt(BytesIO(raws.encode('ascii')), delimiter=[Lf, 1, 1]*hdr['Nobs'])
```
I know np.genfromtxt is handy, but not the fastest. Did you checkout `pandas.read_csv`? According to some sources it is much quicker. Alternatively, a specialized self-written read function may be quickest. I think it is worth spending some extra thoughts on this one, since it is the center-piece of the whole software.

scivision, 2018-08-24:
The current `master` commit has implemented the full ~20x speedup over v1.6.0, which is an additional 2x speedup over v1.6.1, by using a pure Python implementation instead of np.genfromtxt

VOMAY, 2018-08-27:
I see you actually changed quite a lot in the code to achieve this, but you managed to improve the speed by moving away from np.genfromtxt.

Thanks for the update!

closed 2018-08-27 by VOMAY