geospace-code/georinex #35: Read .o files with extreme long time

ZephyrShannon, 2018-11-07:
I am trying to read a  .o file with gr.read(), but it cost me very long time >10m, can any one helps me?

[ista3090.18o.gz](https://github.com/scivision/georinex/files/2557167/ista3090.18o.gz)

ZephyrShannon, 2018-11-07:
sorry, github doesn't allow me upload a .o file so I tar it to gz

ZephyrShannon, 2018-11-07:
[ista3090.18o.gz](https://github.com/scivision/georinex/files/2557204/ista3090.18o.gz)
Sorry, I upload a link file

ZephyrShannon, 2018-11-07:
[ista3090.18o.gz](https://github.com/scivision/georinex/files/2557208/ista3090.18o.gz)
Sorry, I upload a wrong file. this one is ok.
md5sum:6dca94165d45a455c0bb05b1b564ed68

scivision, 2019-03-06:
@ZephyrShannon we recently added the ability to read data from an OBS2 or OBS3 at every `interval=` seconds. For example if the big file has measurements once a second, you can read every 15 seconds or whatever suits your needs, giving a proportion speedup in reading.

The real solution of preallocating Numpy arrays has not yet been implemented for OBS3, but was several months ago for OBS2. I'm sorry for this delay, I just haven't had time to do this yet.

fabprol, 2019-05-10:
Dear scivision, regarding this update, I have found one rinex file that leaves an error of the type:
IndexError(f'may be "fast" mode bug, try fast=False or "-strict" command-line option {e}')
please, find the rinex file at ftp://gssc.esa.int/gnss/data/daily/2013/360/usno3600.13d.Z

Otherwise, if I remove the fast option, by saying fast=False in the function rinexobs2 when it calls the rinexsystem2, the rinex reading becomes correct again.

scivision, 2019-05-10:
Thanks @fabprol for providing the link to the example data, I plan to look at this soon.

fabprol, 2019-05-11:
Hey scivision, indeed, the rinex I sent before is working normally, the one that is not really working is this one: https://drive.google.com/open?id=199UHtvZVkhXG6iMl07bNI4zeMnaR1r5D

which is the same, but converted to RINEX by crx2rnx and edited by Windows/TEQC using the following command: teqc.exe -S -O.obs C1P1P2L1L2 usno3600.13o > usno3600Edited.13o