geospace-code/georinex #37: when reading rinex 3.x file for Galileo an exception occurs, the cause being in the header ionospheric corr when it parses 4 floats where only 3 are available

kasc7, 2019-01-16:
1. There are many types of RINEX files.
     Almost every issue is because of some corner case with a particular RINEX format.
     Thus we will usually need a copy of your RINEX file.
     If it's very large/private, consider copying just the part of it that causes the issue for you.
2. Please copy and paste the error traceback, to help find which function the error is occurring in.
3. generally a dump of all the versions of Python packages you have is **not** needed

kasc7, 2019-01-16:
ValueError: could not convert string to float:

the cause being in the header ionospheric corr when it parses 4 floats where only 3 (a0,a1,a2) are available

kasc7, 2019-01-16:
If the comments are somewhat weird above., sorry for that, I was not familiar with this github s/w.
(probably I cannot delete comments)

To say it again in this one comment:

When I read a rinex 3 nav file for galileo, I get an ValueError exception. because it parses 4 values and only gets 3. It is as of quite current version in nav3.py / navheader3 function in the list comprehension statement in the later part of the function (coefficients = [ .... for i in range(**4**)]

kasc7, 2019-01-18:
I attached the galileo rinex nav file (hope I did it correctly:)
[brdc0320.16l.txt](https://github.com/scivision/georinex/files/2771836/brdc0320.16l.txt)

scivision, 2019-01-18:
Fixed by https://github.com/scivision/georinex/releases/tag/v1.7.1, thanks for uploading the test file

closed 2019-01-18 by scivision