geospace-code/georinex #39: Select signals per GNSS system

alainmuls, 2019-03-04:
Dear Michael,

I have another question. Suppose we have a mixed GPS / Galileo RINEX observation file. Both systems have as observable type C1C and L1C. Now suppose that I only to read the C1C for both systems but only the C1C signal for Galileo. How should I specify the "meas" or do I have to select the same signals for both GNSS systems if they are both available?

kind regards
Alain

scivision, 2019-03-05:
Currently, we can't filter for different measurement sets on different systems.

So now, it's either ['L1C', 'C1C'] for every system you selected, or 'C1C' for every system selected, there isn't a subset selection like that at this time.

To enable this feature, one would have to consider if there would be enough benefit in read speedup. Is the motivation to read the files faster with the suggested subset of measurements?

scivision, 2019-03-05:
One can of course do such selections after reading, that is fundamental to the choice of using `xarray` as an output data type.

alainmuls, 2019-03-06:
Dear Michael,

Thanks for your answer. The intention was to separate the observables of
the systems, but I discovered now also the possibility to just select a
GNSS which does the job for me.

bye/ALain

*Alain Muls*
*Professor*
Department CISS
*Royal Military Academy*
p: +32 2 44 13936
a: Renaissance Avenue 30
  B1000 Brussels (Belgium)
w: www.rma.ac.be/ciss/en/  e: alain.muls@rma.ac.be

alainmuls, 2019-04-04:
Dear Michael,

Yet another request from me. 
I read RINEX obs file in its full length, sometimes up to 24hr @ 1Hz.
Is it possible for the plot command to specify a specific time interval, something like

tlim=['2017-02-23T12:59', '2017-02-23T13:13']

which can be used at reading the RINEX file?

Tx/Alain

alainmuls, 2019-04-04:
Hi Michael,

Sorry, me again.
Since sv and time are both coordinates, I thought it would be something like:

ax.plot(obsData.time, obsData['C1C'].sel(sv=['E04','E01'], time= ['2019-02-10T11:00:00', '2019-02-10T11:20:00']).dropna(dim='time',how='all'))

to select the first 20 minutes of a 1 hour NetCDF file obtained from RINEX obs data. 

Info from that file:
Dimensions:  (sv: 6, time: 3600)
Coordinates:
  * sv       (sv) object 'E01' 'E04' 'E14' 'E19' 'E21' 'E27'
  * time     (time) datetime64[ns] 2019-02-10T11:00:00 ... 2019-02-10T11:59:59

But then I get the error:
KeyError: "not all values found in index 'time'"

I tried with adding ", method='nearest'" but that gives then errors about not being able to substraction or domething of that sort.

Also sometimes (only selecting on sv) the dropna part is giving errors, so when not using it, the plots are made.

Tx/Alain