geospace-code/georinex #41: Can't use the CLI tool after pip install

fabiopicchi, 2019-04-06:
Hi! 

First of all, thank you very much for this great tool. I am already using it from my python scripts but I am not able to use the CLI tools such as ReadRinex.

Looking inside my virtualenv folder I found a **ReadRinex** inside bin and it looked like this:

    #!/Users/fabiopicchi/.venvs/test/bin/python3.7
    # -*- coding: utf-8 -*-
    import re
    import sys
    
    from ReadRinex import main 
    
    if __name__ == '__main__':
        sys.argv[0] = re.sub(r'(-script\.pyw?|\.exe)?$', '', sys.argv[0])
        sys.exit(main())

However, there is no ReadRinex module installed. Under site-packages I just have the **georinex** package. Thus, I get the following error when trying to run this CLI tool:

    Traceback (most recent call last):
        File "/Users/fabiopicchi/.venvs/test/bin/ReadRinex", line 6, in <module>
        from ReadRinex import main
    ModuleNotFoundError: No module named 'ReadRinex'

I have never dug into the pip packaging process so I can't exactly point where the actual problem is (I read the setup.cfg file and everything seems fine). Maybe I am doing something wrong?

Also, I think it would be very useful to provide the Hatanaka compression and decompression tools as binaries. This way, running a project that has georinex as a dependency in different machines will be a little bit more convenient if this feature is required.

scivision, 2019-04-25:
Hi @fabiopicchi I am not sure why this is happening. This install mechanism is tested continuously in https://github.com/scivision/georinex/blob/master/tests/test_scripts.py

On Linux, the installed ReadRinex looks like:
```python
#!/home/username/miniconda3/bin/python
# EASY-INSTALL-ENTRY-SCRIPT: 'georinex','console_scripts','ReadRinex'
__requires__ = 'georinex'
import re
import sys
from pkg_resources import load_entry_point

if __name__ == '__main__':
    sys.argv[0] = re.sub(r'(-script\.pyw?|\.exe)?$', '', sys.argv[0])
    sys.exit(
        load_entry_point('georinex', 'console_scripts', 'ReadRinex')()
```

On Windows the installed ReadRinex looks like:
```python
#!c:\Miniconda3\python.exe
# EASY-INSTALL-ENTRY-SCRIPT: 'georinex','console_scripts','ReadRinex'
__requires__ = 'georinex'
import re
import sys
from pkg_resources import load_entry_point

if __name__ == '__main__':
    sys.argv[0] = re.sub(r'(-script\.pyw?|\.exe)?$', '', sys.argv[0])
    sys.exit(
        load_entry_point('georinex', 'console_scripts', 'ReadRinex')()
    )
```

I only test with MacOS in Travis-CI, but the same ReadRinex passes there too. I am using Python 3.7 with MacOS, Linux and Windows.  Sorry I'm not sure what's happening

dallinbriggs, 2019-04-29:
I am also having this issue on Ubuntu 18.04 using Python 3.6.7. This appears to be a problem with pip not installing it correctly. I had to uninstall georinex and then follow the instructions given for the development version. I had to use sudo to install it. These are the steps I took:
```
git clone https://github.com/scivision/georinex

cd georinex

sudo python3 -m pip install -e .
```