geospace-code/georinex #42: Reading Error for Commented RINEX files - v1.9.0

fabprol, 2019-04-24:
Hey guys, I am using georinex to read some RINEX files from IGS (International GNSS Service). It is working really well and I would like to thank you for providing such a tool. 

However, there are some .obs files with a few comments in the middle of the file (not only in the header). See for instance the following RINEX file, at the following line regarding to the observations at 1 hour:

file: ftp://cddis.gsfc.nasa.gov/gnss/data/daily/2013/359/13o/mana3590.13o.Z
line: 13 12 25  1  0  0.0000000  4  5

You will find  some strange comments, such as:

 13 12 25  1  0  0.0000000  4  5
MANA (COGO code)                                            COMMENT
0.000      (antenna height)                                    COMMENT
+12.14896250 (latitude)                                       COMMENT
-86.24899354 (longitude)                                     COMMENT
0077.169      (elevation)                                        COMMENT

As far as I have detected, georinex is not reading correct such files. It reads the observations bellow the comments and says that such obsearvations are related to the time obtained in the comments plus 30 seconds (or any other interval). However, the next time is also at 1 hour. As a result, georinex is shifting all times when we find this problem.

Indeed, this strange lines are not suposed to be there, so it is a bug when the agency provides the RINEX files. I have seen this kind of comments since many years ago, which are obtained due to the use of TEQC when generating RINEX files. I do not know if this would be easy to solve in georinex, but I think it should be reported.

VOMAY, 2019-04-24:
Hey,
I am part of the IGS community and will report the file. This behavior is highly uncommon. 
I don't see a need to implement this.

If you have any questions concerning IGS your welcome to contact me.

Best Regards
Volker

fabprol, 2019-04-24:
Hello Volker, thanks for the feedback.

This does not only happens with IGS files. I have shown this example because of the free access to the data!

Sorry for the inconvenience.

Regards!

VOMAY, 2019-04-24:
I updated my original answer.

scivision, 2019-04-24:
This bug was fixed in GeoRinex v.1.10.0, Thanks for reporting this issue for a previous GeoRinex release. The fix is to upgrade to latest GeoRinex release.

closed 2019-04-24 by scivision