geospace-code/georinex #43: georinex.read() : Read error for mixed rinex observation files

anand-satnav, 2019-05-03:
Hello Michael,

`dat = gr.load("cutsamp.16o",meas=['C1C','S1C',],use=['G'])
print(dat)`

Output
`Dimensions:  (sv: 12, time: 1) Coordinates:
`

The Dataset returned after the load does not

anand-satnav, 2019-05-03:
I inspected the code and I think I might have found the bug. Please confirm

File           :  georinex/obs3.py 
Function  :  _epoch
Line          :  204 - 209

When a single satellite system is present in hdr['fields'], the elif condition is triggered calling xarray.concat. Concatenation of data and epoch_data results in an empty dataset since data is empty. This is surprising as I expected it to concatenate all 'epoch_data' information into 'data'

When I remove this elif condition, and re-run the problem is solved . The else condition is triggered calling xarray.merge which correctly merges data and epoch_data

scivision, 2019-05-17:
Thanks for the detailed report, we'll look into this

scivision, 2019-09-03:
Sorry for my delay. Using your file:
```sh
dat = gr.load("cutsamp.16o",meas=['C1C','S1C',],use=['G'])
```
results in

```
<xarray.Dataset>
Dimensions:  (sv: 12, time: 1)
Coordinates:
  * time     (time) datetime64[ns] 2016-01-08
  * sv       (sv) <U3 'G18' 'G07' 'G15' 'G17' 'G20' ... 'G30' 'G05' 'G13' 'G19'
Data variables:
    C1C      (time, sv) float64 2.519e+07 2.588e+07 ... 2.053e+07 2.23e+07
    S1C      (time, sv) float64 35.8 33.0 48.2 49.6 43.2 ... 43.5 42.1 49.9 49.0
Attributes:
    version:            3.02
    interval:           30.0
    rinextype:          obs
    fast_processing:    0
    time_system:        GPS
    filename:           cutsamp.16o
    position:           [-2364337.2971, 4870285.5843, -3360809.8188]
    position_geodetic:  (-32.003888575142994, 115.89479894212391, 24.00298790...
```

I wonder if something has been fixed in xarray or georinex in the meanwhile. I just tested this with georinex git revision 666b950