geospace-code/georinex #46: Take measures by a range date after read

rodolfolotte, 2019-05-20:
Well, after to read a lot, I could not find something that allows me to restrict the measures by provide a range of two dates. The reason to do that, is that my rinex is really heavy. and the .load() would be too unnecessary! 

So, when read by the first time, I can set the full-day reading, like:
```
obs = gr.load(complete_path, meas=columns, use=constellations)
```
and if I want a specific range, I would just use 'tlim' tag. Ok, so, after to get my `obs`, I want to get all measures that is related to a specific date, without to read it again. So, I've tried:
```
obs_short = obs.sel(time=[initial_date, final_date])
```
but `obs_short` will be only a two measures variable, not a range! I also have tried `tlim` in the .sel():
```
obs_short = obs.sel(tlim=[initial_date, final_date])
```
but there is no such property in `.sel`! 

Someone could help?

scivision, 2019-05-20:
Thanks for reaching out. So it looks like you want to index an existing xarray.Dataset after reading.
The way I would do this is with a boolean indexer, something like
```python
t = obs.time
i = (t >= initial_date) & (t < final_date)
obs_short = obs.isel(i)
```

I didn't get a chance to try it now, but the syntax would look like that

rodolfolotte, 2019-05-21:
Oh, thank you so much! 

Ok, I have read about the index, but I thought it could only be used for columns! So, I just tried your recommendation out and got: ` '>=' not supported between instances of 'int' and 'datetime.datetime'`. Actually, the `obs.time` is a `xarray.DataArray` with string timestamps, I'm still searching how to compare!

rodolfolotte, 2019-05-30:
@scivision I'm sorry, buddy! Could you help me to sort it out? I have tried both way:
```
range_option_1 = initial_date <= obs.time <= final_date  
range_option_2 = settings.INITIAL_HOUR_RECALC_BIAS <= obs.time <= settings.FINAL_HOUR_RECALC_BIAS  
obs_reduced = obs.isel(range_range_option_2)  
```
but still, did not work!