geospace-code/georinex #48: Enhacement: Allow several messages per epoch and satellite on Galileo depending on Data source

PabloPerdi, 2019-06-21:
Broadcasted navigation RINEX can contain several navigation messages on the same epoch depending on navigation message decoded for Galileo (INAV[E1B, E5b] / FNAV [E5a]) in a particular station (signal decoded). It would be great to allow the xarray to contain several ephemeris per satellite on the same epoch to differenciate between data sources.

daniestevez, 2019-07-21:
I needed this functionality when doing [this study](https://destevez.net/2019/07/galileo-operational-again/), so I hacked together a [modified version of georinex](https://github.com/daniestevez/georinex/tree/rinexnav_nonunique) that allows several navigation messages for the same SVN and timestamp.

The way this is done is that additional SVN entries are created in the xarray `Dataset` to store the additional entries (which often come from different data sources). These are numbered with a suffix counter. Thus, you get E01, E01_1, E01_2, ..., each of these storing the data for different data sources.

Maybe it would be good to include this behaviour into georinex so that it can be toggled as an option when reading navigation RINEX files. My modification only covers RINEX 3.

eballes, 2019-07-23:
I'm having a similar problem for GPS and RINEX 2 where I need to gather data from multiple stations and use the 'TransTime' parameter.

I think another possibility would be to dimension the xarray using three coordinates. Now we are currently using the `toc` and the `sv`. Adding an extra parameter (e.g. `TransTime`) will make possible the `xarray.merge` operation without having to lose data.

scivision, 2019-09-03:
OK thanks for this interesting approach and commentary.
I noticed so many of the NAV times have duplicate entries, and for the work we were doing at the time, we simply discarded the duplicates, but I thought there must be some cases for which the discarded data is useful.

So is simply appending an arbitrary integer as @daniestevez suitable for the purposes here?

scivision, 2019-09-03:
This was merged in #52, please let us know if anything further can be improved.

closed 2019-09-03 by scivision