geospace-code/georinex #64: Microseconds Error in sp3 Reader

Shaylah-Maria, 2020-03-23:
The time in the file is:
2018  9 23  0  2 37.63292462

Georinex is saving datetime as:
2018-9-23T00:02:37.063292  (incorrect)

instead of:
2018-9-23T00:02:37.63292  (correct)

See first data point entry:
[image]

What is stored inside georinex.to_datetime(data.time[ii]) (first element of the saved dates in sp3 file). this should equal the first date point in file above:
[image]

To fix, must multiply what is saved in microseconds by 10.

-Shaylah Mutschler

scivision, 2020-06-03:
Thank you I seem to have overlooked this. I have flagged this so I can examine and fix. Thanks for your patience and kindly feedback.

mauriciodev, 2021-09-30:
It seems that line 145 of sp3.py should have "ln[23:29]" instead of "ln[23:28]"

        microsecond=int(ln[23:29]),

commit 746ccdc9 (2021-11-26): bugfix: sp3 microsecond

fixes #64
commit 34ca76e6 (2021-11-26): bugfix: sp3 microsecond
fixes #64
commit d66a1a97 (2021-11-26): bugfix: sp3 microsecond

fixes #64
closed 2021-11-26 by scivision