geospace-code/georinex #66: Exception while reading observation file

shuri, 2020-08-31:
1. The RINEX file is this one: https://rnl.ae.utexas.edu/texcup/2019May09-rover/asterx4/SEPT1291.19O
    (9M)
2. traceback for the code 
    gr.load(SEPTENTRIO_ASTERX4_DATA)
---------------------------------------------------------------------------
IndexError                                Traceback (most recent call last)
<ipython-input-40-d57e77b742f2> in <module>
----> 1 OBSERVABLES = gr.load(SEPTENTRIO_ASTERX4_DATA)
      2 #OBSERVABLES = gr.rinexheader(SEPTENTRIO_ASTERX4_DATA)

~/h/virtualenvs/i/lib/python3.8/site-packages/georinex/base.py in load(rinexfn, out, use, tlim, useindicators, meas, verbose, overwrite, fast, interval)
     64                         overwrite=overwrite)
     65     elif info['rinextype'] == 'obs':
---> 66         return rinexobs(rinexfn, outfn, use=use, tlim=tlim,
     67                         useindicators=useindicators, meas=meas,
     68                         verbose=verbose,

~/h/virtualenvs/i/lib/python3.8/site-packages/georinex/base.py in rinexobs(fn, outfn, use, group, tlim, useindicators, meas, verbose, overwrite, fast, interval)
    184 
    185     if int(info['version']) in (1, 2):
--> 186         obs = rinexobs2(fn, use, tlim=tlim,
    187                         useindicators=useindicators, meas=meas,
    188                         verbose=verbose,

~/h/virtualenvs/i/lib/python3.8/site-packages/georinex/obs2.py in rinexobs2(fn, use, tlim, useindicators, meas, verbose, fast, interval)
     36     attrs: Dict[str, Any] = {}
     37     for u in use:
---> 38         o = rinexsystem2(fn, system=u, tlim=tlim,
     39                          useindicators=useindicators, meas=meas,
     40                          verbose=verbose,

~/h/virtualenvs/i/lib/python3.8/site-packages/georinex/obs2.py in rinexsystem2(fn, system, tlim, useindicators, meas, verbose, fast, interval)
    231                         data[i*3+2, j, isv] = darr[:, k*3+2]
    232                 else:
--> 233                     data[i, j, isv] = darr[:, k]
    234 # %% output gathering
    235     data = data[:, :times.size, :]  # trims down for unneeded preallocated

IndexError: index 57 is out of bounds for axis 2 with size 36

3. Python version is 
    3.8.2 (v3.8.2:7b3ab5921f, Feb 24 2020, 17:52:18) 
    [Clang 6.0 (clang-600.0.57)]
    (running inside jupyter)

remzawi, 2020-09-22:
I have the same problem with another file from the same dataset.

remzawi, 2020-09-23:
Looking at the code and the observation files I guess the problem comes form "Nsvsys = 36" line 94 of obs2.py, becausesome satellites have a higher number than 36. Could it be possible to increase it perhaps?

scivision, 2020-09-23:
Thank you very much for these detailed reports!