geospace-code/georinex #70: 60 seconds

labonocl, 2020-10-19:
Hi there,

One of the sp3 files for ESA's Swarm mission (ftp://swarm-diss.eo.esa.int/Level1b/Latest_baselines/GPSxNAV/, Satellite A, file for November 26th 2013) contains the following line:

*  2013 11 26  0  1 60.00000000

When trying to read this line with load_sp3, the datetime library complains with the following error:

ValueError: second must be in 0..59

The error is of course correct (this one file alone contains this error 64 times), so I see two options here:
1) Put on my Don-Quijote-hat and try to convince ESA that this is an error and that they should debug their data product processing chain and reprocess the entire seven years of Swarm data; or
2) check for such errors in sp3dt, and fix them there.

Now, I know what I am going to do. So the real question becomes: would you be interested in pulling such a correction of wrong times into the georinex package, or is your position that whoever is responsible for producing sp3 files should take care of this issue, and it is not the responsibility of georinex to handle other people's mistake?

Cheers
Lasse

labonocl, 2020-10-19:
Having spend more time looking into this, the entire sp3dt method seems wrong. According to the sp3-c definition (which is the current one, as sp3-a was discontinued in 2006 - although the format of this line has not changed), an epoch header has the format:
```
SP3 Line Twenty three (The Epoch Header Record)

     Columns 1-2         Symbols            *_                  A2
     Column  3           Unused             _                   blank
     Columns 4-7         Year Start         2001                I4
     Column  8           Unused             _                   blank
     Columns 9-10        Month Start        _8                  I2
     Column  11          Unused             _                   blank
     Columns 12-13       Day of Month St    _8                  I2
     Column  14          Unused             _                   blank
     Columns 15-16       Hour Start         _0                  I2
     Column  17          Unused             _                   blank
     Columns 18-19       Minute Start       _0                  I2
     Column  20          Unused             _                   blank
     Columns 21-31       Second Start       _0.00000000         F11.8
```
However, sp3dt does the following:
```
def sp3dt(ln: str) -> datetime:
    return datetime(
        int(ln[3:7]),
        int(ln[8:10]),
        int(ln[11:13]),
        int(ln[14:16]),
        int(ln[17:19]),
        int(ln[20:22]),
        int(ln[23:28]),
    )
```

scivision, 2020-10-21:
I think this is something georinex should tolerate. There are all kinds of quirks in RINEX that are non standard compliant

commit dbc259d2 (2020-11-11): fix sp3 with wrapping time

fixes #70
closed 2020-11-11 by scivision