geospace-code/georinex #71: Error when loading high-rate data

danielecini, 2020-11-30:
High-rate data are usually available as Hatanaka-compressed files:
https://cddis.nasa.gov/archive/gnss/data/highrate/2020/288/20d/00/
As I decompress one of these files, turning it into a readable observation file (for instance, gamb288a00.20o), I get an error by simply trying to load the file; in particular, the error is usually of this type:

 File "...\Python\Python39\lib\site-packages\georinex\obs2.py", line 233, in rinexsystem2
    data[i, j, isv] = darr[:, k]
IndexError: index 37 is out of bounds for axis 2 with size 36

In some previous issues it has been reported that the error may arise from line 94 of obs2.py; nevertheless, even if I change that value (Nsvsys = 36) the error message does not change. Moreover, for _certain_ high-rate observation files georinex works perfectly, so I guess there is some counting issue that I cannot detect. As of now, I've been using georinex for quite some months and I just started using it for high-rate data: I wonder if high-rate files differ in some crucial aspect, making them more likely to fail.

Python version: Python 3.9.0 64bit

scivision, 2021-11-28:
Is there a specific file I can try? The hatanaka implementation was recently change to use the Python hatanaka library.

shradhamohanty, 2022-11-14:
Hello, even I have the same issue while using gr.load(filename)
```
[/usr/local/lib/python3.7/dist-packages/georinex/obs2.py](https://localhost:8080/#) in rinexsystem2(fn, system, tlim, useindicators, meas, verbose, fast, interval)
    247                         data[i * 3 + 2, j, isv] = darr[:, k * 3 + 2]
    248                 else:
--> 249                     data[i, j, isv] = darr[:, k]
    250     # %% output gathering
    251     data = data[:, : times.size, :]  # trims down for unneeded preallocated

IndexError: index 39 is out of bounds for axis 2 with size 36
```
I am unable to attach the sample obs file because of compatibility issue with file type and even zipping the file is too large. But I too have a high rate (1Hz) obs file that wont load using georinex.

Thanks and regards,
Shradha