geospace-code/georinex #88: gr.load() error on RINEX 2 observation file

pietrovermicelli, 2022-07-25:
1. RINEX OBS 2.11 2.11. 
     File: kaza0150.22o.
3. 
Traceback (most recent call last):
  File "/Users/pietrovermicelli/Desktop/firstpythonproject/RINEX.py", line 5, in <module>
    dat = gr.load("/Users/pietrovermicelli/SpacEarth Dropbox/SpacEarth Dropbox/calibrazione_TEC_pietrovermicelli/abpo0210.16o")
  File "/Users/pietrovermicelli/.local/share/virtualenvs/firstpythonproject-q2IRJTKJ/lib/python3.10/site-packages/georinex/base.py", line 67, in load
    return rinexobs(
  File "/Users/pietrovermicelli/.local/share/virtualenvs/firstpythonproject-q2IRJTKJ/lib/python3.10/site-packages/georinex/base.py", line 217, in rinexobs
    obs = rinexobs2(
  File "/Users/pietrovermicelli/.local/share/virtualenvs/firstpythonproject-q2IRJTKJ/lib/python3.10/site-packages/georinex/obs2.py", line 54, in rinexobs2
    obs = xarray.merge((obs, o))
  File "/Users/pietrovermicelli/.local/share/virtualenvs/firstpythonproject-q2IRJTKJ/lib/python3.10/site-packages/xarray/core/merge.py", line 1023, in merge
    merge_result = merge_core(
  File "/Users/pietrovermicelli/.local/share/virtualenvs/firstpythonproject-q2IRJTKJ/lib/python3.10/site-packages/xarray/core/merge.py", line 752, in merge_core
    aligned = deep_align(
  File "/Users/pietrovermicelli/.local/share/virtualenvs/firstpythonproject-q2IRJTKJ/lib/python3.10/site-packages/xarray/core/alignment.py", line 827, in deep_align
    aligned = align(
  File "/Users/pietrovermicelli/.local/share/virtualenvs/firstpythonproject-q2IRJTKJ/lib/python3.10/site-packages/xarray/core/alignment.py", line 764, in align
    aligner.align()
  File "/Users/pietrovermicelli/.local/share/virtualenvs/firstpythonproject-q2IRJTKJ/lib/python3.10/site-packages/xarray/core/alignment.py", line 551, in align
    self.align_indexes()
  File "/Users/pietrovermicelli/.local/share/virtualenvs/firstpythonproject-q2IRJTKJ/lib/python3.10/site-packages/xarray/core/alignment.py", line 404, in align_indexes
    joined_index = joiner(matching_indexes)
  File "/Users/pietrovermicelli/.local/share/virtualenvs/firstpythonproject-q2IRJTKJ/lib/python3.10/site-packages/xarray/core/indexes.py", line 442, in join
    coord_dtype = np.result_type(self.coord_dtype, other.coord_dtype)
  File "<__array_function__ internals>", line 180, in result_type
TypeError: The DType <class 'numpy.dtype[float64]'> could not be promoted by <class 'numpy.dtype[datetime64]'>. This means that no common DType exists for the given inputs. For example they cannot be stored in a single array unless the dtype is `object`. The full list of DTypes is: (<class 'numpy.dtype[float64]'>, <class 'numpy.dtype[datetime64]'>)
4. the gr.load() used to work just fine until Friday on this observation file, and other Rinex 2.11 obs. However, today I opened my project and I got the error above. I don't get the error while loading the corresponding nav file.

m-montaqami, 2022-11-11:
please use the following versions:
pandas == 0.25.3
numpy  == 1.19.0
xarray == 0.12.1

pietrovermicelli, 2022-11-16:
Thank you, I was using differen versions. Although I've found that by enforcing the dtype to be 'object' for coord_type, I don't get that error anymore, and it doesn't seem to compromise the correct reading of the observations.

gleatd01, 2022-12-06:
> Thank you, I was using differen versions. Although I've found that by enforcing the dtype to be 'object' for coord_type, I don't get that error anymore, and it doesn't seem to compromise the correct reading of the observations.

Hello, I am relatively new to numpy, how are you enforcing the dtype to be object?

pietrovermicelli, 2022-12-06:
Hi, if you're getting the same error, I think I did something of the kind:
try:
     coord_dtype = np.result_type(self.coord_dtype, other.coord_dtype)
except:
     coord_dtype = np.dtype('object') #here I am enforcing the type 'object' in case the result_type fails
more on numpy.dtype https://numpy.org/doc/stable/reference/generated/numpy.dtype.html

gleatd01, 2022-12-07:
I ended up using python 3.8 and the above 3 packages. it did work for loading rinex files.