geospace-code/georinex #104: GPS L2 is empty

clabo16, 2024-03-20:
I tested observation files from different stations, both rinex 2 and 3. All the time GPS L2 is empty while e.g. GPS L1 and Glonass L1 and L2 data is there.

An example file I am using is ftp://epncb.eu/pub/obs/2015/076/KIR800SWE_U_20150760000_01D_30S_MO.crx.gz
My Code:
```
L2_all=georinex.load('KIRU00SWE_R_20150760000_01D_30S_MO.crx.gz', use='G',meas='L2C')
print('L2 GPS: ', L2_all)
L2_all=georinex.load('KIRU00SWE_R_20150760000_01D_30S_MO.crx.gz', use='R',meas='L2C')
print('L2 Glonass: ', L2_all)
```
Result:

```
L2 GPS:  <xarray.Dataset> Size: 140B
Dimensions:  (time: 1, sv: 11)
Coordinates:
  * time     (time) datetime64[ns] 8B 2015-03-17T23:59:30
  * sv       (sv) <U3 132B 'G18' 'G13' 'G19' 'G05' ... 'G27' 'G21' 'G07' 'G16'
Data variables:
    *empty*
Attributes:
    version:          3.02
    interval:         30.0
    rinextype:        obs
    fast_processing:  0
    time_system:      GPS
    filename:         KIRU00SWE_R_20150760000_01D_30S_MO.crx.gz
    position:         [2251420.932, 862817.14, 5885476.604]
L2 Glonass:  <xarray.Dataset> Size: 576kB
Dimensions:  (time: 2880, sv: 24)
Coordinates:
  * time     (time) datetime64[ns] 23kB 2015-03-17 ... 2015-03-17T23:59:30
  * sv       (sv) <U3 288B 'R01' 'R02' 'R03' 'R04' ... 'R21' 'R22' 'R23' 'R24'
Data variables:
    L2C      (time, sv) float64 553kB nan nan nan ... 8.083e+07 8.475e+07
Attributes:
    version:          3.02
    interval:         30.0
    rinextype:        obs
    fast_processing:  0
    time_system:      GPS
    filename:         KIRU00SWE_R_20150760000_01D_30S_MO.crx.gz
    position:         [2251420.932, 862817.14, 5885476.604]
```

I ran the codes with python 3.11.6 and 3.8.10

HiddenTurtleAI, 2024-03-20:
Hi, check the rinex header first:
hdr = georinex.rinexheader('ASCG00SHN_R_20233000000_01D_30S_MO.crx.gz')
When I used your code on my rinex
file: ASCG00SHN_R_20233000000_01D_30S_MO.crx.g  it was also amepty. But
when I checked the header (hdr['fields']['G']) I saw that there are L2
observations, but in diffrent modulation: L2W and L2X for gps and L2P/L2C
for Glonass (hdr['fields']['R']). I'm working with double frequency data
for all the time with georinex so I'm pretty sure it works

clabo16, 2024-03-21:
Thank you very much for your quick help. Indeed, there is data in L2L and L2W.

closed 2024-03-21 by clabo16